Time-Appliances-Project/Time-Card #10: BOM and schematic errata

wisxxx, 2021-08-13:
**BOM**

- item 41 (DS3231 RTC) shows "U5" as ref des. schematic shows "U3A"

- item 40 (1x3 header) does not have corresponding jumpers

- no BOM item for CSAC and SA.3x/SA.5x

- no BOM item for U11 (sch. p.6,A2; assume SiT5721)

- no BOM item for U13 (sch. p.6,B4; assume SiT5356)

- no BOM item for U4 (sch. p.5,C5; BMP388. Note: this part is not recommended for new designs)

- no BOM item for U7 (sch. p.5,D5; AT24MAC402-STUM-T)

**SCHEMATIC**

- p. 6,C2

    MAC devices SA.3x and SA.5x are labeled "RV-3049-C3"

    Need notation: "Install one of CSAC, SA.3x, or SA.5x".

    Assuming connector U10B would interfere with installation of the CSAC or
    SA.3x oscillators, need notation: "Install U10B (connector) only if using SA.5x oscillator."

ahmadexp, 2021-08-14:
Love it. Thank you so much. I will go through the correction list.

wisxxx, 2021-09-20:
It appears the the analog frequency control input to the MAC-SA5x oscillator is connected to a digital I/O pin in the FPGA. Is this intended?

In 'Timingcard_SCHEMATICS_BETA_V1.pdf", p. 6,A3 and p. 7,B4, the MAC_FREQ_CTRL net connects the MAC-SA5x INPUT FREQUENCY CONTROL pin (#1) to the FPGA board CON2-B33.

According to the AC7100B FPGA board schematic ("AC7100B_SDM.pdf"), CON2-B33 is routed to Xilinx pin AA10, which is listed in the XC7A100T pinout ("ug475_7Series_Pkg_Pinout.pdf") as a DQS (multifunction) pin.

I assume that analog frequency control is not enabled in the MAC-SA5x. Whatever, the wiring probably needs an update.

AlphaBetaPhi, 2021-09-20:
@wisxxx, this is correct. The analog tuning feature is not used and all tuning is performed through the digital interface. 

On the FPGA this pin is not used either, it's a placeholder is for alternative versions should they arise in R&D.

closed 2023-01-30 by wisxxx