Time-Appliances-Project/Time-Card #11: What are the IMU (BNO080) and barometric pressure sensor (BMP388) used for?

Nufflee, 2021-08-13:
As the title says, I'm curious what these sensors are used for in a time measurement device, furthermore because they don't seem to be mentioned anywhere outside of the ECAD.

ahmadexp, 2021-08-14:
The environmental and IMU sensors are used to monitor the surrounding condition since that directly affects the atomic clock especially when it is under holdover mode. Any changes or disturbances should be considered as a factor that increases the uncertainty.

Nufflee, 2021-08-14:
I see, thank you. Would you mind expanding on which specific environmental factors would increase the uncertainity of the system? Since this is made for datacenters I can't imagine a lot going on other than maybe some subtle vibrations and temperature/pressure changes.

ahmadexp, 2021-08-14:
That's right. Keep in mind, I designed the card fof general purpose application including DCs (as the first one). The datasheet of the SA53 talks about various environmental factors and their effect on the clock stability

closed 2021-08-18 by AlphaBetaPhi

wisxxx, 2021-09-17:
The MAC-SA5x data sheet (Microchip #DS00003348B, March 2021) only mentions temperature in terms of operating range and frequency drift:

  • Operating Temperature –40˚C to +75˚C
  • Temperature-induced frequency errors <5x10-11 Hz/Hz from –10˚C to +75˚C
  • Non-Operating (Storage & Transport):  –55˚C to +100˚C

Shock and Vibration are only specified with respect to maintaining lock.

Altitude (pressure) is only mentioned as a maximum for operation and storage/transport.

Can anyone explain how oscillator performance can be compensated for environmental effects based on what the spec sheet is presenting?

If loss of lock from temperature or shock/vibration occurs, won't other environmental monitoring in a data center be able to report? If localized over-temperature is a concern, then perhaps only that should be measured.