Time-Appliances-Project/Time-Card #16: Is this absolute hard limit->PCIe x1 (18 pins) generation 3.0 or above

oiaohm, 2021-08-16:
There is a reason why is this a hard limit or is it that lower has not been tried.   If this card can work with generation 2.0/2.1 this opens a few other hardware options.

Raspberry Pi 4 Compute Module supporting boards can provide PCI 2.0 and RK3399 based boards can provide PCI 2.1.

Basically a 1U rack being able to hold primary secondary clock sync sources and few other extras. 

For the Raspberry PI 4 Compute Module I am sure that "Jeff Geerling"
https://www.youtube.com/channel/UCR-DXc1voovS8nhAvccRZhg
would try the card.

Smaller setups its possible that something like a Raspberry PI 4 compute module would have enough processing power for like 50 machines worth of clock sync.    Think shop point of sales system.

wisxxx, 2021-08-18:
It would be electrically compatible (by PCIe definition), but the performance decrease would likely affect system operation.

It may be that there is an application where reduced accuracy, latency, and jitter is acceptable (e.g., a low cost time source). For such, it might be difficult to get the system validation under this project, given the time appliance's intended audience (time lords).

ahmadexp, 2021-08-18:
[image]

Time Card has no problem to run on PCIe x1. As a matter of fact, I tested it on the CM4. Jeff Geerling has a Time Card en-route. I am excited as you waiting for Jeff's magic with the Time Card.

oiaohm, 2021-08-18:
I think by that test the Readme can update.
https://github.com/opencomputeproject/Time-Appliance-Project/blob/master/Time-Card/README.md

PCIe x1 (18 pins) generation 3.0 or above << this comes a recommend for accuracy, latency, and jitter.   I would say test min could come 2.0.

Working on a CM4 board means it works with PCIe generation 2.0  the CM4 is a cut down PCIe 2.0 as it missing parts the specification says should be there.   Rockchip PCIe 2.1 will have no problem if a CM4 work.

I have a few different usage cases having a clock source that can run locally and not drift badly will be useful.   We have not had broken internet or jammed GPU for while now but its happened in the past.

https://centerclick.com/ntp/
I see this card as a better option to lot these GPS backed ones.   From what I can see this time card would be a more stable time source if GPS is disrupted.   Yes the atomic clock costs.   This is a rock pi S board based system as RK3308 CPU yes it cpu weaker than the cm4 as well.

There are a lot cases where the GPS time server in the network so some small arm based system.   Horrible even if they are built from open source supporting parts parts their software stack is common closed.   I have never felt happy having a time server that I cannot have audited at the source code level due to how much trouble a miss behaving time server can cause.

The facebook issue with closed source time source being a worry don't just apply to server farms.   It does apply to like pos setups where you want to keep the pos machines and the security system aligned for one.    Pos machines information aligned to bank processing information is also important that your clock has not drift too far.    Lower accuracy and jitter than data centres need but you cannot drifting by too much either.

Also looking at the card it looks like you could join 2 time card to each other by the PPS ports.   This could be useful in different setups.   So that cases like security system and pos systems on two different networks are in fact working on the same time with no general data link.   This is something I cannot do with the entry class gps time systems.   This also makes me wonder about a primary/secondary setup.   As in primary card has GPS and atomic lock and secondary card is only the fpga the job here would be to make the same time on two independent networks.

I know I am looking for same level accuracy that facebook is but I still need reasonable about of accuracy and precision in a lot of these cases  and have the problems when you have gps distruptions.   Of course for profit the more places the time card can be sold the better.

ahmadexp, 2021-08-19:
I am honored to see you passion for this project.

ahmadexp, 2021-08-19:
Jeff just received a Time Card and is going to do his magic to get it work on the CM4:
 https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/199