Time-Appliances-Project/Time-Card #17: FPGA IP Licensing

wisxxx, 2021-08-17:
What are the licensing or royalty arrangements for the FPGA intellectual property?

There appear to be six IP cores from NetTimeLogic in use.

ahmadexp, 2021-08-18:
The current license with NetTimeLogic allows the use of the bin file. 
Alternatively we are working on a version without the proprietary IPs. Hopefully coming soon.

thschaub, 2021-08-20:
Exactly, it is allowed to use the IP cores as implemented in the bin file. If you have adaptation proposals which have a common interest we can also check if it make sense to adapt something as part of this binary license.

There are five NetTimeLogic IP Cores as part of the binary license included (PPS Slave, PPS Master, TOD Slave, Adjustable Clock and Signal Timestamper). The "UART to AXI" and "Static config" are included in the licensing of any other IP Core.
Additionally, there are IP Cores included from Xilinx (AXI UART, AXI Memory Mapped to PCI Express, AXI IIC, AXI GPIO etc.) which are under their end user license agreement:
https://www.xilinx.com/content/dam/xilinx/licenses/ip/end-user-license-agreement.pdf

Currently we are checking different options for a version without the proprietary IP.

geerlingguy, 2021-09-20:
@AlphaBetaPhi - Was there any other update on this or is this issue closed as other options won't be investigated at this time?

AlphaBetaPhi, 2021-09-20:
Hi @geerlingguy, we are still investigating this. I'll label appropriately.

thschaub, 2022-03-10:
There will be soon an update about the open source FPGA version.

https://www.opencompute.org/wiki/Time_Appliances_Project
on Mar-23, 2022, there will be a call about Open Time Card FPGA

wisxxx, 2022-06-03:
Thanks. I see the pinout in 

 

…/Implementation/Xilinx/TimeCard/Constraints/PinoutConstraint.xdc

 

Good info.

 

From: Thomas Schaub ***@***.***> 
Sent: Friday, June 3, 2022 5:20 AM
To: opencomputeproject/Time-Appliance-Project ***@***.***>
Cc: wisxxx ***@***.***>; Mention ***@***.***>
Subject: Re: [opencomputeproject/Time-Appliance-Project] FPGA IP Licensing (#17)

 

Hi @geerlingguy <https://github.com/geerlingguy> , hi @wisxxx <https://github.com/wisxxx> 
a first version of the base system (without TOD Slave and PPS Slave) is released:
https://github.com/opencomputeproject/Time-Appliance-Project/tree/master/Time-Card/FPGA/Open-Source

wisxxx, 2022-06-07:
The pinout appears to show that the PCie interface is reduced to one lane. Am I reading it correctly?

thschaub, 2022-07-01:
> The pinout appears to show that the PCie interface is reduced to one lane. Am I reading it correctly?

Not really related to this topic. 
If you build the open source FPGA project you could see the configuration of the PCIe memory mapped bridge in Vivado. And yes it's only one lane.

thschaub, 2023-01-26:
Full Open-Source Project including video tutorials available:
https://github.com/opencomputeproject/Time-Appliance-Project/tree/master/Time-Card/FPGA/

closed 2023-01-26 by thschaub