Time-Appliances-Project/Time-Card #18: Unknown connector

wisxxx, 2021-08-18:
What is this external connector? It isn't in the schematic, but seems to be associated with the GNSS receiver.

[image]

ahmadexp, 2021-08-18:
That is a simple RF proto module which allows you to build all kinds of filters, attenuators, etc... 
At the moment I have a pass through (just a wire) inside it.
This was place there because we are planning to use an active antenna (GPS-OVER-FIBER) and that will require an attenuator (-30db) on the line to avoid receiver [input]([image]) saturation.

ahmadexp, 2021-08-18:
[image]

wisxxx, 2021-08-18:
Thanks. It seems then that I could use a simple SMA/SMA (or SMB/SMB) pass-through connector if I have a powered patch antenna a run of low-loss cable.

ahmadexp, 2021-08-18:
That's right. I have done that too. 
[image]

closed 2021-08-18 by AlphaBetaPhi