Time-Appliances-Project/Time-Card #23: Driver compilation on Raspberry Pi (CM4) / Pi OS

geerlingguy, 2021-08-20:
I'm working on getting this thing working on a Raspberry Pi, running the latest version of Raspberry Pi OS:

```
$ uname -a
Linux cm4 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux

$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Right now the `remake` script fails with:

```
$ ./remake 
Could not locate kernel source for distribution 'debiangnu/linux'
```

But if I update the script to allow debian in addition to ubuntu:

```diff
diff --git a/Time-Card/DRV/remake b/Time-Card/DRV/remake
index 61e0c73..77ed189 100755
--- a/Time-Card/DRV/remake
+++ b/Time-Card/DRV/remake
@@ -10,7 +10,7 @@ if [ "$*" == "install" ]; then
 fi
 
 case "$DIST" in
-    ubuntu)
+    debian*|ubuntu)
         KDIR=/usr/src/linux-headers-${KVER}
         ;;
     centos*)
```

Then I get:

```
$ ./remake 
make: Entering directory '/usr/src/linux-headers-5.10.52-v8+'
  CC [M]  /home/pi/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.o
/home/pi/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.c:956:22: warning: ‘struct firmware’ declared inside parameter list will not be visible outside of this definition or declaration
         const struct firmware *fw)
                      ^~~~~~~~
/home/pi/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.c: In function ‘ptp_ocp_devlink_flash’:
/home/pi/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.c:968:12: error: dereferencing pointer to incomplete type ‘const struct firmware’
  resid = fw->size;
            ^~
/home/pi/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.c: In function ‘ptp_ocp_devlink_flash_update’:
/home/pi/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.c:1012:50: error: ‘struct devlink_flash_update_params’ has no member named ‘fw’
  err = ptp_ocp_devlink_flash(devlink, dev, params->fw);
                                                  ^~
make[1]: *** [scripts/Makefile.build:280: /home/pi/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.o] Error 1
make: *** [Makefile:1824: /home/pi/Time-Appliance-Project/Time-Card/DRV] Error 2
make: Leaving directory '/usr/src/linux-headers-5.10.52-v8+'
```

geerlingguy, 2021-08-20:
@philhartung mentioned [here](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/199#issuecomment-902946392) that it may be because Pi OS ships with a 5.10 kernel, whereas 5.11 or newer is required for the module to build. I'm now testing compiling in the driver from kernel source on 5.14... we'll see how it goes (see issue mention above this comment).

ahmadexp, 2021-08-20:
I speculate that is due to using a older version of gcc (the built-in for Pi vs the one on CentOS (C++11))
@jlemon, any insight on this?

ATR83-05, 2021-09-04:
Egnnn

closed 2021-09-21 by ahmadexp