Time-Appliances-Project/Time-Card #24: PCIe Bracket V3 dimensions

wisxxx, 2021-08-20:
The vertical hole centers are referenced to the top surface of the short bend section at the upper end of the bracket. This position, being on the outside of the bend, is not as well controlled as the bottom surface. The CEM and off-the-shelf brackets all use the bottom surface as a reference.

The horizontal hole centers _are_ properly referenced to the inside of the bend of the ears.

Also, not specified are:
- the overall length of the bracket
- the tapered section on the lower end
- the widths and lengths of the mounting tabs
- the width of the stiffener opposite the mounting tabs

wisxxx, 2021-09-17:
SMA connectors do not appear centered in the cutouts when the .STP model is combined with a 3D model of the connectors (Amphenol 901-143-6RFX, from SnapEDA) mounted on a 1.57 mm (61.8 mil) thick board. The offset is estimated to be about 0.64 mm.

The connector model dimensions appear match the part drawings. Both models are very close to their correct positions vertically. Mounting the bracket on the top side of of the board puts the connectors below the center of the cutouts.

It may be that the cutouts are oversized to accommodate different SMA connectors. Is that the case?

looking from outside:
[image]

zoom
[image]

mounting detail
[image]

ahmadexp, 2021-09-17:
@wisxxx, I really want to thank you for all the amazing reviews you have been doing on the time card.

wisxxx, 2023-01-30:
Addressed in bracket V4 docs.