Time-Appliances-Project/Time-Card #32: Accelerometer and Pressure Sensors

wisxxx, 2021-09-17:
In the now-closed issue #11, the requirements for acceleration and pressure sensor were laid to the MAC-5x oscillator data sheet. However...

The MAC-SA5x data sheet (Microchip #DS00003348B, March 2021) only mentions temperature in terms of operating / non-operating range, and frequency drift:

• Operating Temperature –40˚C to +75˚C
• Temperature-induced frequency errors <5x10-11 Hz/Hz from –10˚C to +75˚C
• Non-Opera

ting (Storage & Transport): –55˚C to +100˚C

Shock and Vibration are only specified with respect to maintaining lock.

Altitude (pressure) is only mentioned as a maximum for operation and storage/transport.

Can anyone explain how oscillator performance can be compensated for environmental effects based on what the spec sheet is presenting?

If loss of lock from temperature or shock/vibration occurs, won't other environmental monitoring in a data center be able to report? If localized over-temperature is a concern, then perhaps only that should be measured.

Are there any other citations for these environment impacts that are relevant to this application?

AlphaBetaPhi, 2021-09-29:
Hi @wisxxx,

The purpose of these sensors is to collect environmental information local to the card for purposes of analyzing the data trends over longer periods of time. We report all data over the PCIe driver and can be used for determining effects of temperature, vibration, shock, and aging in cases where holdover is critical.

closed 2021-09-29 by AlphaBetaPhi

wisxxx, 2021-09-29:
Does this drive the placement of the sensor then? To capture the effect on the oscillator, it seems it would ideally be placed on the back side of the board, centered in the oscillator footprint. That would capture the Z axis motion most effectively - a simple sign change (if needed), compared to extrapolation from some other position on the board (along with the resonant mode analysis required to make it worthwhile).