Time-Appliances-Project/Time-Card #34: EEPROM write protect input driven by +1.5V signal

wisxxx, 2021-09-21:


wisxxx, 2021-09-21:
Looking at "Timingcard_SCHEMATICS_BETA_V1.pdf", the AT24MAC402 serial EEPROM WP input (p. 4,A5) is driven by the EXT_EEPROM_WP signal from the FPGA board (pin P1A-44, p. 6,B5) and a 10K pullup to +3.3V.

The AT24MAC402 is operating on +3.3V. VIL min is specified as VCC x 0.7 = +2.31 V.

The FPGA pin driving the signal to P1A-A44 (net B34_L21_P) is driven by Bank 34, which is a +1.5V block.

- Does the 10k pullup to +3.3V guarantee a valid logic high into the EEPROM? It seems to depend on a very large output resistance in the FPGA buffer, given the 10K pullup value.
- Can the FPGA output reliably sustain the back-drive (nominal 180 uA)?

ref: AC7100 FPGA SoM schematic "AC7100B_SDM.pdf" (p. 6,D2; p.12,D4), "AC7100B User Manual.pdf".

AlphaBetaPhi, 2021-09-21:
Hi @wisxxx, 

Thank you for a great issue.

This is correct. The WP has been moved to a 3.3V bank in the next version we are evaluating.

wisxxx, 2022-06-04:
EEPROM write protect is +1.5V signal in new FPGA description

File: PinoutConstraint.xdc, lInes: 75-77

```
#EEPROM
set_property PACKAGE_PIN V9 [get_ports EepromWp_DatOut]
set_property IOSTANDARD **LVCMOS15** [get_ports EepromWp_DatOut]
```

wisxxx, 2023-01-13:
**Not fixed** in file: Time-Appliance-Project/blob/master/Time-Card/FPGA/Open-Source/Implementation/Xilinx/**TimeCard**/Constraints/PinoutConstraint.xdc

**Is fixed** (moved to pin AA13) in file: Time-Appliance-Project/blob/master/Time-Card/FPGA/Open-Source/Implementation/Xilinx/**TimeCard_Production**/Constraints/PinoutConstraint.xdc

thschaub, 2023-01-13:
Just to clarify: The constraint files are for two different hardware versions. That's why they are different. The FPGA pinout constraints are simply configured that it matches with the hardware. So you will never see a "fix" in the constraints file since they are related to a specific hardware.

wisxxx, 2023-01-30:
Fixed in recent FPGA pushes.

closed 2023-01-30 by wisxxx