Time-Appliances-Project/Time-Card #38: Updated FPGA

wisxxx, 2021-09-29:
"Time-Card/SOM/FPGA/TimeCard_FPGA.png" shows an extra GNSS receiver (UART and PPS connections) as well as an UART in the upper left that is not labelled. These don't appear on the SOM connector P1 of the current schematic ("Timingcard_PROJECT_SCHEMATICS_BETA_V1.pdf", p.7). This change is reflected in the FPGA binaries directory as well.

- Are they being resurrected from the initial design shown in "Time-Card/ECAD/Prototype-V0/Timingcard_PROJECT.pdf"?
- Or, are they simply no-connects on P1?
- Or, can we expect a new design with an FPGA replacing the AC7100 SOM? If so, when would that land?

AlphaBetaPhi, 2021-09-29:
Hi @wisxxx,

We are currently evaluating a set of configuration variants for the Timecard which we plan to release in a few weeks. Additionally, there will be support for backward compatibility.

thschaub, 2021-12-10:
We try to keep backwards compatibility of the FPGA image where ever possible. The base functionality should work on all Hardware revisions.
So it might be that you see in the FPGA more or different features which are not available on a specific HW revision. 
The FPGA image provides the features as in the description but it is possible that the base board Timecard does not map the pins from the SOM. 

The not labled UART (Communication UART to access the AXI Adddress Space) is answer in #39

closed 2023-01-30 by wisxxx