Time-Appliances-Project/Time-Card #44: I found a mistake I think

wlong-coxeve, 2021-12-09:
1. "Time-Card\SOM\FPGA\Readme.pdf" shows that ANT1 and ANT2 are input ports; 
However, "Time-Card\ECAD\Beta-V1\USER_IO.SchDoc" shows that ANT1 and ANT2 are output ports;
Futhermore, i can only get 10MHz (from my crystal oscillator) on the 4th pin of the "U8"(NC7SZ125M5), and there is nothing output on all four SMA ports ".

Q1: Could you please explain the contradiction?

2. “Time-Card\GNSS\UBX\RCB-F9T\README.md" recommands me to program the GNSS module by the UBX U-Center, but the 9th generation GNSS is not supported for the function of programming by U-Center.

Q2: Is it necessary for me to only get PPS on timecard port?
Q3: Does the RCB-F9T blong to 9th generation GNSS ? If so, How can I resolve it？

Last but not least, all these problems are based on the fact that I did not plug in PCIe, but only the external 12V power supply.

Thank you so much !!!

thschaub, 2021-12-09:
**Q1:** Your observation is in general correct. The schematics you are reffering does not match with the FPGA readme. The schematics is a beta (should we replace this once? @ahmadexp) and it was wrong because with this you loose the backwards compatibility. Do you have produced your own PCB? 
On the [picture ](https://github.com/opencomputeproject/Time-Appliance-Project/blob/master/Time-Card/images/timingcard.png) you can see that it was patched on the hardware.

So in short the FPGA/Readme.pdf is correct.

**2.** In the u-center I am using 9th generation is supported (Version 20.06.01)

**Q2:** I don't understand what you exactly mean by this?

**Q3:** I guess so but as mentioned I don't see there any problem with U-Center

The FPGA comes up with a base configuration, however without PCIe or UART you can't do any modification or reading out any time (I guess the schematics you are reffering to doesn't have an FPGA UART). I would recommand to continue with the PCIe and the driver.

wlong-coxeve, 2021-12-09:
I am so  glad to receive your replay!  Thank you!  lol !!!
Q1: Solved;
Q2:
The first step I only testing the basic features of FPGA, 10MHz and PPS，so that insure  my own PCB has not faulty. But there is no PPS output, so i think whether i need to further programming the GNSS (or something) according to the README.

Q3: Solved;

thschaub, 2021-12-10:
**Q2:** Theoretically you should see a PPS even without a GNSS module. By default on ANT4 the PPS of the Local Clock from the PPS master is selected.
Additionally you can maybe check some status on the LEDs. On LED3 you should get also a PPS. They have following mapping (Readme.pdf is updated with this information):
LED1: Alive LED of the FPGA internal Clock (50MHz) --> This must be blinking!
LED2: Alive LED of the PCIe clocking part (62.5MHz)
LED3: PPS of the FPGA (Time of the Local Clock via PPS Master)
LED4: PPS of the MAC (differential inputs from the MAC via diff-buffer)

wlong-coxeve, 2021-12-10:
> **Q2:** Theoretically you should see a PPS even without a GNSS module. By default on ANT4 the PPS of the Local Clock from the PPS master is selected. Additionally you can maybe check some status on [...]

wisxxx, 2021-12-13:
The marked up schematic page 1, to reflect the assembly:

[image]