Time-Appliances-Project/Time-Card #46: FPGA pin assignments for production version

wisxxx, 2022-01-18:
What are the pin assignments for the "Version: 0x8003 (Production Version)" FPGA?

They aren't in "Readme_Production.pdf".

thschaub, 2022-03-10:
It's not planned to have the pin assignment in the "Readme_Production.pdf" and the schematic (and PCB) of this timecard version is currently not published. Therefore no pinout available at the moment.

ahmadexp, 2022-03-16:
@wisxxx, please try to join our upcoming meeting about opening the FPGA code of the time card
https://www.linkedin.com/feed/update/urn:li:activity:6909908144374652929

thschaub, 2023-01-26:
Pinout constraint file for the Open Source Production version is available here:
https://github.com/opencomputeproject/Time-Appliance-Project/blob/master/Time-Card/FPGA/Open-Source/Implementation/Xilinx/TimeCard_Production/Constraints/PinoutConstraint.xdc

closed 2023-01-26 by thschaub