Time-Appliances-Project/Time-Card #61: question How does this interface with pcie?

Nosh-Ware, 2022-06-02:
I was trying to do some research on how to make some diy electronics that use PCIe, and after a fruitless search baring the slot's pinout, I ended up here, being the only relay active project using PCIe (not Mini PCIe). How exactly does this, on the electronic level, interface with PCIe? Is that something documented elsewhere? Is this something I should ask somewhere else?

ahmadexp, 2022-06-20:
If I understood your question correctly, the PCIe interface is via the FPGA and a core from AXI that provides a memory controller. In the Memory controller was have a set of addresses assigned for various things which in turn the Linux driver maps them to the operating system in different forms such of POSIX clock, etc…
The PCIe connection itself is basically made from two pairs of differential signals (one clock and one data for a 1X connection). Please do not hesitate ask further questions and I apologize in advance if I did not get your question right.

Nosh-Ware, 2022-09-08:
Sorry about the long delay here, but yes that was a lot of helpful information, thanks! I do have a few other questions though. 

I'm hoping to have some type of microcontroller (preferably an Arduino Uno/ATMega328P, but I can understand if the speeds warrant a more powerful or otherwise different choice) interface with the slot for some datalogging of my own (temperature, humidity, and 6dof sensors, not anything as complex as you have here). I understand how its a lot more reasonable to use USB to do all of this but for the sake of experimentation, lets assume I'm the unreasonable type. Also I do know about some PCI mini slots having a embedded USB interface as well, but I still consider that "cheating".

I was wondering if there's any good resources I missed about how to make a electronic device interface with PCIe that's freely available (ie not requiring me to be hired by a company registered with the people who maintain the standard). Preferably one as simple as possible (I am not necessarily a complete beginner, but something as advanced as this will no doubt still be a lot to figure out).