Time-Appliances-Project/Time-Card #66: Some questions about MAC and Address mapped

Satoshier, 2022-07-21:
1. I  noticed that MAC your official recommend is SA53. Can I use some other MAC or OCXO ,TCXO which can offer 10MHZ frequency source?  If possible，how to deal with the problem of pins which are mismatch with the PCB of Beta-V1.  I just wanna to replace it with our company’s MAC

2. You opened source the implementation of FPGA then I imported the Vivado project. According the TC_SmaSelector IP CORE , I can configure the SMA input/ouput ports. If I change the configuration, does it means that I should regenerate the bitstream files? I think it is a bit of trouble.    In /Time Appliance Project/Time Card/Usage, by using the LINUX command such as " echo OUT: MAC >> sma1" which can also configure the ports.  Both of them are equivalent？？？

3.  In /SOM/FPGA/TestApp/TestAPP.c, I can only read result form the register, how do I write data to the register?

Looking forward your reply, THX.

Satoshier, 2022-07-21:
@thschaub  @leoleovich

julianstj1, 2023-01-25:
1. You can use different MACs , OCXOs, and TCXOs. We will release an OCXO daughter card that sits in the same footprint as the SA53 to support this soon. Please take a look at the latest v7 board, it has some options to support different types of MACs.
2. I personally use the echo OUT: MAC >> sma1 method to configure ports and it works great. 
3. Unsure of this one, I haven't used the testapp.

thschaub, 2023-01-26:
2. The idea of the SMA selector is that you can change them on run time via the registers. So no there is no regeneration of the bitstream required if you would like to change the mapping.

3. The TestApp.c uses /dev/mem and mmap and then basic pointer operations in C.

Here an example how you can do a write:
```
virt_addr = (char*)pcie_base + NTL_CLK_REGSET_BASE+NTL_CLK_CLKCONTROL_REG;
 write_value = NTL_CLK_CONTROL_ENABLE_BIT | NTL_CLK_CONTROL_TIME_ADJ_BIT;
*(volatile uint32_t*)virt_addr = write_value;
```