Time-Appliances-Project/Time-Card #80: Fail in creating open source FPGA binaries from scratch

wisxxx, 2023-01-11:
I'm trying to build the open source FPGA binaries following the video "How to build the open source FPGA design for the Time Card from scratch", linked in the FPGA "README.md" file.

The first step, cloning the repository, was successful. The "CreateProject.tcl" script in Vivado appears to fail.  I'm using the latest version of Vivado:

	Vivado v2022.2 (64-bit)
	SW Build: 3671981 on Fri Oct 14 05:00:03 MDT 2022
	IP Build: 3669848 on Fri Oct 14 08:30:02 MDT 2022

The TCL console shows:

```
start_gui
source C:/Users/wis1/Documents/tc/Time-Appliance-Project/Time-Card/FPGA/Open-Source/Implementation/Xilinx/TimeCard/CreateProject.tcl
# set ScriptFile [file normalize [info script]]
# set ScriptFolder [file dirname $ScriptFile]
# cd $ScriptFolder
# set origin_dir $ScriptFolder
# set orig_proj_dir "[file normalize "$origin_dir"]"
# set _xil_proj_name_ "TimeCard"
# create_project ${_xil_proj_name_} ./${_xil_proj_name_} -part xc7a100tfgg484-1 -force
INFO: [ProjectBase 1-489] The host OS only allows 260 characters in a normal path. The project is stored in a path with more than 80 characters. If you experience issues with IP, Block Designs, or files not being found, please consider moving the project to a location with a shorter path. Alternately consider using the OS subst command to map part of the path to a drive letter.
Current project path is 'C:/Users/wis1/Documents/tc/Time-Appliance-Project/Time-Card/FPGA/Open-Source/Implementation/Xilinx/TimeCard/TimeCard'
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository 'C:/Xilinx/Vivado/2022.2/data/ip'.
# set proj_dir [get_property directory [current_project]]
# set obj [current_project]
# set_property -name "default_lib" -value "xil_defaultlib" -objects $obj
# set_property -name "dsa.accelerator_binary_content" -value "bitstream" -objects $obj
WARNING: [Common 17-599] Property 'dsa.accelerator_binary_content' is deprecated for object type 'project'. dsa.* properties have been deprecated, please use the corresponding platform.* properties.
# set_property -name "dsa.accelerator_binary_format" -value "xclbin2" -objects $obj
WARNING: [Common 17-599] Property 'dsa.accelerator_binary_format' is deprecated for object type 'project'. dsa.* properties have been deprecated, please use the corresponding platform.* properties.
# set_property -name "dsa.description" -value "Vivado generated DSA" -objects $obj
WARNING: [Common 17-599] Property 'dsa.description' is deprecated for object type 'project'. dsa.* properties have been deprecated, please use the corresponding platform.* properties.
# set_property -name "dsa.dr_bd_base_address" -value "0" -objects $obj
WARNING: [Common 17-599] Property 'dsa.dr_bd_base_address' is deprecated for object type 'project'. dsa.* properties have been deprecated, please use the corresponding platform.* properties.
# set_property -name "dsa.emu_dir" -value "emu" -objects $obj
ERROR: [Common 17-142] Invalid property name 'dsa.emu_dir'.  Please use the 'list_property' command to find properties supported by the target 'project' object.
```

The Messages tab shows:

```
General Messages[IP_Flow 19-234] Refreshing IP repositories
[IP_Flow 19-1704] No user IP repositories specified
[IP_Flow 19-2313] Loaded Vivado IP repository 'C:/Xilinx/Vivado/2022.2/data/ip'.
[Common 17-599] Property 'dsa.accelerator_binary_content' is deprecated for object type 'project'. dsa.* properties have been deprecated, please use the corresponding platform.* properties.
[Common 17-142] Invalid property name 'dsa.emu_dir'.  Please use the 'list_property' command to find properties supported by the target 'project' object.
```

This is the statement in "CreateProject.tcl" (line 51) which appears to cause the problem:

`	set_property -name "dsa.emu_dir" -value "emu" -objects $obj
`

Other than loading Vivado 2019.4, used in the video, what can I modify to get this working?

thschaub, 2023-01-12:
We are aware that the script is not compatible with newer Vivado versions. As mentioned in the Video @ ~2:18 we are sticking at the moment to **Vivado 2019.1**. The link to download Vivado 2019.1 you can find in the video notes.
I will create soon a video tutorial how to upgrade to newer Vivado Versions.

As soon the video is ready you will find it in the [Tutorial List](https://github.com/opencomputeproject/Time-Appliance-Project/tree/master/Time-Card/FPGA)

closed 2023-01-12 by thschaub

thschaub, 2023-01-13:
@wisxxx: Some minor updates of the scripts are available which are a base for newer Vivado versions https://github.com/opencomputeproject/Time-Appliance-Project/commit/154b1fbb3b0fa988b60b91651199859bf8e32e80. There will be some more adaptations required which I will explain in a new tutorial.

wisxxx, 2023-01-13:
Successful build with Vivado 2019.1 of the "TimeCard". I saw errors in "TimeCard_Production" - is that expected?

wisxxx, 2023-01-13:
Could not repeat errors on "TimeCard_Production". It is now built successfully.

thschaub, 2023-01-17:
@wisxxx a new video tutorial is available which describes the changes we made to be Vivado version independent https://www.youtube.com/watch?v=lijj4WYPaPA