Time-Appliances-Project/Time-Card #81: Open source FPGA build fails

wisxxx, 2023-01-12:
After installing Vivado 2019.1 and cloning the time card repository to my local machine, the script "CreateBinariesAll.tcl" fails with this error:
```
ERROR: [BD 41-758] The following clock pins are not connected to a valid clock source: 
/TC_PpsSlave_1/SysClk_ClkIn
/TC_PpsSlave_1/SysClkNx_ClkIn

ERROR: [BD 41-1031] Hdl Generation failed for the IP Integrator design C:/Users/wis1/Documents/tc/Time-Appliance-Project/Time-Card/FPGA/Open-Source/Implementation/Xilinx/TimeCard/TimeCard/TimeCard.srcs/sources_1/bd/TimeCard/TimeCard.bd 
ERROR: [Vivado 12-4756] Launch of runs aborted due to earlier errors while preparing sub-designs for run execution.

    while executing
"source "$ScriptFolder/CreateBinariesGolden.tcl""
    (file "C:/Users/wis1/Documents/tc/Time-Appliance-Project/Time-Card/FPGA/Open-Source/Implementation/Xilinx/TimeCard/CreateBinariesAll.tcl" line 28)
```

There are warnings in the Tcl console during "CreateProject.tcl", the first of which is:
```
## create_root_design ""
INFO: [IP_Flow 19-5107] Inferred bus interface 'ClkIn0_ClkIn' of definition 'xilinx.com:signal:clock:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'ClkIn1_ClkIn' of definition 'xilinx.com:signal:clock:1.0' (from Xilinx Repository).
WARNING: [IP_Flow 19-3153] Bus Interface 'ClkIn0_ClkIn': ASSOCIATED_BUSIF bus parameter is missing.
WARNING: [IP_Flow 19-3153] Bus Interface 'ClkIn1_ClkIn': ASSOCIATED_BUSIF bus parameter is missing.
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository 'c:/Users/wis1/Documents/tc/Time-Appliance-Project/Time-Card/FPGA/Open-Source'.
WARNING: [IP_Flow 19-3656] If you move the project, the path for repository 'c:/Users/wis1/Documents/tc/Time-Appliance-Project/Time-Card/FPGA/Open-Source' may become invalid. A better location for the repository would be in a path adjacent to the project. (Current project location is 'c:/Users/wis1/Documents/tc/Time-Appliance-Project/Time-Card/FPGA/Open-Source/Implementation/Xilinx/TimeCard/TimeCard'.)
WARNING: [BD 41-1731] Type mismatch between connected pins: /BufgMux_IPI_0/ClkOut_ClkOut(undef) and /BufgMux_IPI_2/ClkIn0_ClkIn(clk)
WARNING: [BD 41-1731] Type mismatch between connected pins: /BufgMux_IPI_1/ClkOut_ClkOut(undef) and /BufgMux_IPI_2/ClkIn1_ClkIn(clk)
WARNING: [BD 41-1731] Type mismatch between connected pins: /BufgMux_IPI_2/ClkOut_ClkOut(undef) and /clk_wiz_1/clk_in1(clk)
WARNING: [BD 41-1306] The connection to interface pin /util_ds_buf_1/IBUF_DS_N is being overridden by the user. This pin will not be connected as a part of interface connection CLK_IN_D
    (_repeats 25 times, different instances_)
WARNING: [BD 41-702] Propagation TCL tries to overwrite USER strength parameter S_AXI_SUPPORTS_NARROW_BURST(false) on '/axi_pcie_0' with propagated value(true). Command ignored
```

Subsequently, there are 63 of these warnings(same warning, different blocks and domains):
```
WARNING: [BD 41-927] Following properties on pin /BufgMux_IPI_0/ClkIn0_ClkIn have been updated from connected ip. They may not be synchronized with cell properties. You can set property on pin directly to confirm the value and resolve the warning.
	CLK_DOMAIN=TimeCard_Mhz10ClkSma_ClkIn 
```

thschaub, 2023-01-13:
The warnings you have reported are no issue, it just mentioned that there is no BUS assigned to a clock which is in this case correct since there is only a clock mux. Additionally it does not detect the output as a Clk Ouputs but this should be also no issue.

It's more interessting that you have an Tc_PpsSlave_1 instance in your Block Design. The BD tcl only creates only the [Tc_PpsSlave_0](https://github.com/opencomputeproject/Time-Appliance-Project/blob/31ae07f60544251b1d1306149210d0638d4de711/Time-Card/FPGA/Open-Source/Implementation/Xilinx/TimeCard/Bd/TimeCardBd.tcl#L447): 

**The Tc_PpsSlave_1 should not exist.** Have you touched somehow the block design manually? Did you really created the design from scratch? **(Git Cleanup and remove untracked files before call the CreateProject script)**.

There should be no existing Vivado project folder: Time-Appliance-Project\Time-Card\FPGA\Open-Source\Implementation\Xilinx\TimeCard\TimeCard

wisxxx, 2023-01-13:
I did not knowingly touch anything. I’ll re-clone and try again. 

 

Thanks for this and all the other help.

 

-wis

closed 2023-01-13 by wisxxx