Time-Appliances-Project/Time-Card #90: Production Timecard: bad termination on FPGA JTAG with USB

wisxxx, 2023-03-01:
Ref. schematic R4006-G0001-03-SC-REV02.pdf

If using the USB interface (FT432, sht. 24), the FPGA JTAG signals are double-terminated by resistors shown at sht. 24, zone B11:

> [image]

and sht. 10, zone J4.

> [image]

by way of the mux'ing shown on sht. 24 "JTAG/SPI_MASTER_SEL".

ahmadexp, 2023-08-09:
@julianstj1 did we address this?

julianstj1, 2023-08-09:
This is on the production card, and the USB interface worked when we tested it, but definitely this double termination should not be there. If we ever have another BOM change on there, I will add this on, thanks for the great catch!