Time-Appliances-Project/Time-Card #101: Depreciated function use in 'ptp_ocp' driver

HugoFolcher, 2023-09-11:
For recent kernel versions: (Here with 6.2.0-32)

```
user@user:~/Projects/Time-Appliance-Project/Time-Card/DRV$ sudo bash remake
make : on entre dans le répertoire « /usr/src/linux-headers-6.2.0-32-generic »
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  You are using:           gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  CC [M]  /home/user/Projects/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.o
/home/user/Projects/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.c: In function ‘ptp_ocp_devlink_info_get’:
/home/user/Projects/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.c:1736:15: error: implicit declaration of function ‘devlink_info_driver_name_put’; did you mean ‘devlink_info_serial_number_put’? [-Werror=implicit-function-declaration]
 1736 |         err = devlink_info_driver_name_put(req, KBUILD_MODNAME);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |               devlink_info_serial_number_put
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:260 : /home/user/Projects/Time-Appliance-Project/Time-Card/DRV/ptp_ocp.o] Erreur 1
make: *** [Makefile:2026 : /home/user/Projects/Time-Appliance-Project/Time-Card/DRV] Erreur 2
make : on quitte le répertoire « /usr/src/linux-headers-6.2.0-32-generic »
```
It seems that 'devlink_info_serial_number_put' is no longer available in last kernel.

I fixed this (quick and dirty) by ignoring the driver_name error i.e:
`err = devlink_info_driver_name_put(req, KBUILD_MODNAME);` -> `err = 0;`

ahmetcanfakili, 2023-10-26:
Yes, I have the same problem.

ahmetcanfakili, 2023-10-26:
What is the reason why it does not work in the latest kernel version? Does it cause problems in the operation of TimeCard?

vvfedorenko, 2023-10-26:
the reason is actually because the latest kernel version have the latest driver, it was fully upstreamed a year ago. Linux 6.2+ has all the features updated and fully operational. And if we are not talking about experiments, all user should use upstreamed versions

HugoFolcher, 2023-10-31:
I am aware of this @vvfedorenko, I was asking because I modify the driver for specific uses of the art card (experiments indeed).

closed 2023-11-07 by vvfedorenko