Time-Appliances-Project/Time-Card #104: Should the version numbers in the CoreList be the same as the version registers?

JohnHay, 2023-11-18:
Should the version number in the CoreList match the number in the Version register of that core? There are some that are different. 

```
timecard0: CoreList 0x10000, register 0x10001, TC SMA Selector Inst0 Slave1
timecard0: CoreList  0x10000, register 0x10001, TC SMA Selector Inst0 Slave2
timecard0: CoreList  0x20001, register 0x20002, TC ToD Slave
```

I looks like TodSlave.vhd has a different version number (0x20003) from the above 2 ToD version numbers:
```
constant TodSlaveMajorVersion_Con               : std_logic_vector(7 downto 0) := std_logic_vector(to_unsigned(0, 8));
constant TodSlaveMinorVersion_Con               : std_logic_vector(7 downto 0) := std_logic_vector(to_unsigned(2, 8));
constant TodSlaveBuildVersion_Con               : std_logic_vector(15 downto 0) := std_logic_vector(to_unsigned(3, 16));
```

thschaub, 2023-11-21:
Yes, that's the idea but at the moment it's manual work and therefore the potential high to miss the update in the CoreList as well.
It's fixed in a903680d5155aa2600549b9dd2f26e3a086021e7

JohnHay, 2023-11-21:
Thank you. :-)

Currently my driver compare them in the attach routine and print a message to the console if there is a difference.

JohnHay, 2023-11-23:
I have now loaded the version 9 LitePcie firmware and still see a version mismatch on the SMA Selector. It is not an issue for me, so you do not have to bring out a new version just for that. I just thought I'll mention it for when you next bring out a new version:

```
timecard0: CoreList 0x10000, register 0x10001, TC SMA Selector Inst0 Slave1
timecard0: CoreList 0x10000, register 0x10001, TC SMA Selector Inst0 Slave2
```