Time-Appliances-Project/Time-Card #106: Verifying GPS

SfCal, 2024-03-30:
I'm experiencing issues as I'm following along the Time Card installation and usage guide. 

OS: Ubuntu 22.04.4 LTS x86_64
Host: NUC9VXQNX K47173-402
Kernel: 5.15.0-101-generic

Being on Ubuntu 22.04.4 I went straight to step 11 and made sure the driver was up to date.

I'm a bit concerned about missing NMEA in dmesg here

```
[    0.486218] pci 0000:02:00.0: [1d9b:0400] type 00 class 0x058000
[    0.486242] pci 0000:02:00.0: reg 0x10: [mem 0x90000000-0x91ffffff]
[    0.602696] pci 0000:02:00.0: Adding to iommu group 1
[    5.205701] ptp_ocp 0000:02:00.0: enabling device (0000 -> 0002)
[    5.207805] 0000:02:00.0: ttyS5 at MMIO 0x90161000 (irq = 175, base_baud = 3125000) is a 16550A
[    5.210203] 0000:02:00.0: ttyS6 at MMIO 0x90181000 (irq = 177, base_baud = 3125000) is a 16550A
[    5.212064] ptp_ocp 0000:02:00.0: Time: 1711770702.739673435, UNSYNCED
[    5.212124] ptp_ocp 0000:02:00.0: Version 1.2.0, clock PPS, device ptp3
[    5.212128] ptp_ocp 0000:02:00.0: TOD Version 2.2.1
[    5.212135] ptp_ocp 0000:02:00.0: control: 10000001
[    5.212137] ptp_ocp 0000:02:00.0: TOD Protocol UBX enabled
[    5.212138] ptp_ocp 0000:02:00.0: GNSS ALL
[    5.212142] ptp_ocp 0000:02:00.0: status: 0
[    5.212146] ptp_ocp 0000:02:00.0: correction: 0
[    5.212149] ptp_ocp 0000:02:00.0: utc_status: 0
[    5.212150] ptp_ocp 0000:02:00.0: utc_offset: 0  valid:0  leap_valid:0
[    5.212154] ptp_ocp 0000:02:00.0: version 1a
[    5.212155] ptp_ocp 0000:02:00.0: regular image, version 26
[    5.212156] ptp_ocp 0000:02:00.0: GNSS @ /dev/ttyS5 115200
[    5.212157] ptp_ocp 0000:02:00.0: MAC @ /dev/ttyS6   57600
```

Here's the output from cgps 
```
lqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqklqqqqqqqqqqqqqqqqqqSeen  0/Used  0k
x Time:          n/a (0)                    xxGNSS   PRN  Elev   Azim   SNR Usex
x Latitude:        n/a                      xx                                 x
x Longitude:       n/a                      xx                                 x
x Alt (HAE, MSL):        n/a,       n/a     xx                                 x
x Speed:           n/a                      xx                                 x
x Track (true, var):                n/a deg xx                                 x
x Climb:           n/a                      xx                                 x
x Status:         NO FIX (0 secs)           xx                                 x
x Long Err  (XDOP, EPX):  n/a ,  n/a        xx                                 x
x Lat Err   (YDOP, EPY):  n/a ,  n/a        xx                                 x
x Alt Err   (VDOP, EPV):  n/a ,  n/a        xx                                 x
x 2D Err    (HDOP, CEP):  n/a ,  n/a        xx                                 x
x 3D Err    (PDOP, SEP):  n/a ,  n/a        xx                                 x
x Time Err  (TDOP):       n/a               xx                                 x
x Geo Err   (GDOP):       n/a               xx                                 x
x ECEF X, VX:              n/a    n/a       xx                                 x
x ECEF Y, VY:              n/a    n/a       xx                                 x
x ECEF Z, VZ:              n/a    n/a       xx                                 x
x Speed Err (EPS):        n/a               xx                                 x
x Track Err (EPD):        n/a               xx                                 x
x Time offset:            n/a               xx                                 x
x Grid Square:            n/a               xx                                 x
mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqjmqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj

{"class":"VERSION","release":"3.22","rev":"3.22","proto_major":3,"proto_minor":14}
{"class":"DEVICES","devices":[]}
{"class":"WATCH","enable":true,"json":true,"nmea":false,"raw":0,"scaled":false,"timing":false,"split24":false,"pps":false}
```

I didn't feel confident that I understood how to use gpsd correctly so I also confirmed with pygpsclient below. I am getting data from ttyS5 but it doesn't seem correct. 

[image]

I purchased the Time Card from Makerfab and used a NUC 9 Pro because it was within the supported systems list. I didn't see any recommendation for a specific GPS receiver so I found a generic one on amazon. It is outside so it has LOS to the sky. Curious what I may be missing here.

SfCal, 2024-04-04:
Wanted to leave an update, switched over to a desktop system I had:

OS: Pop!_OS 22.04 LTS x86_64 
Host: Z690 UD AX DDR4 
Kernel: 6.8.0-76060800daily20240311-generic 

I was able to get gpsd working, I was unable to use the command `gpsd /dev/ttyS5` to set the device and had to instead edit it directly from `/etc/default/gpsd` and restart the service.

[image]

Curiously, I am getting `Could not find device ttySX` but each device appears to be working as expected. Will attempt BIOS update on NUC later.

```
[    1.194851] pci 0000:06:00.0: [1d9b:0400] type 00 class 0x058000 PCIe Endpoint
[    1.194878] pci 0000:06:00.0: BAR 0 [mem 0x42000000-0x43ffffff]
[    1.194972] pci 0000:06:00.0: enabling Extended Tags
[    1.254669] pci 0000:06:00.0: Adding to iommu group 19
[    7.633255] ptp_ocp 0000:06:00.0: enabling device (0000 -> 0002)
[    7.660873] 0000:06:00.0: ttyS4 at MMIO 0x42161000 (irq = 160, base_baud = 3125000) is a 16550A
[    7.674818] 0000:06:00.0: ttyS5 at MMIO 0x42171000 (irq = 161, base_baud = 3125000) is a 16550A
[    7.678954] 0000:06:00.0: ttyS6 at MMIO 0x42181000 (irq = 162, base_baud = 3125000) is a 16550A
[    7.680243] 0000:06:00.0: ttyS7 at MMIO 0x42191000 (irq = 167, base_baud = 3125000) is a 16550A
[    7.681502] ptp_ocp 0000:06:00.0: Could not find device ttyS4
[    7.681504] ptp_ocp 0000:06:00.0: Could not find device ttyS5
[    7.681504] ptp_ocp 0000:06:00.0: Could not find device ttyS6
[    7.681505] ptp_ocp 0000:06:00.0: Could not find device ttyS7
[    7.681522] ptp_ocp 0000:06:00.0: Version 1.2.0, clock PPS, device ptp0
[    7.681533] ptp_ocp 0000:06:00.0: Time: 1712200239.97249309, in-sync
[    7.681534] ptp_ocp 0000:06:00.0:  GNSS: /dev/ttyS4  @ 115200
[    7.681535] ptp_ocp 0000:06:00.0: GNSS2: /dev/ttyS5  @ 115200
[    7.681535] ptp_ocp 0000:06:00.0:   MAC: /dev/ttyS6  @  57600
[    7.681539] ptp_ocp 0000:06:00.0:  NMEA: /dev/ttyS7  @ 115200
```

SfCal, 2024-05-10:
Switched to the newly released Ubuntu 24.04 for some more testing. 

Worked the same as Pop_OS but decided to recompile the kernel anyway in case. The only kernel parameter that wasn't already set correctly as defined by the install guide was `CONFIG_IKCONFIG=y` however this didn't seem to change anything. 

After updating the card firmware this is what I'm getting.

```[    2.330759] ptp_ocp: loading out-of-tree module taints kernel.
[    2.331098] ptp_ocp: module verification failed: signature and/or required key missing - tainting kernel
[    2.332021] ptp_ocp 0000:02:00.0: enabling device (0000 -> 0002)
[    2.332598] ptp_ocp 0000:02:00.0: MSI/MSI-X info: 32
[    2.341284] ptp_ocp 0000:02:00.0: Could not find device ttyS6
[    2.341738] ptp_ocp 0000:02:00.0: Could not find device ttyS7
[    2.341739] ptp_ocp 0000:02:00.0: Could not find device ttyS8
[    2.341740] ptp_ocp 0000:02:00.0: Could not find device ttyS9
[    2.341812] ptp_ocp 0000:02:00.0: Version 1.2.0, clock PPS, device ptp3
[    2.342764] ptp_ocp 0000:02:00.0: Time: 1715354082.175616378, in-sync
[    2.342765] ptp_ocp 0000:02:00.0:  GNSS: /dev/ttyS6  @ 115200
[    2.342787] ptp_ocp 0000:02:00.0: GNSS2: /dev/ttyS7  @ 115200
[    2.342788] ptp_ocp 0000:02:00.0:   MAC: /dev/ttyS8  @  57600
[    2.342791] ptp_ocp 0000:02:00.0:  NMEA: /dev/ttyS9  @   9600
```

SfCal, 2024-06-11:
You'll have to forgive me here as I'm not super familiar with device drivers. 

"Could not find device ttySX is located in `ptp_ocp_link_child()` which itself calls `device_find_child_by_name()`. 

Here's where we see an issue. Printing child devices does not seem to return any of the ttySX's. What is the expected output here?

Here is what I modified in ptp_ocp.c and the corresponding dmesg output.

```
[    0.436299] pci 0000:02:00.0: [1d9b:0400] type 00 class 0x058000 PCIe Endpoint
[    0.436323] pci 0000:02:00.0: BAR 0 [mem 0x90000000-0x91ffffff]
[    0.536194] pci 0000:02:00.0: Adding to iommu group 2
[    3.178399] ptp_ocp: loading out-of-tree module taints kernel.
[    3.178403] ptp_ocp: module verification failed: signature and/or required key missing - tainting kernel
[    3.179012] ptp_ocp 0000:02:00.0: enabling device (0000 -> 0002)
[    3.179298] ptp_ocp 0000:02:00.0: MSI/MSI-X info: 32
[    3.179584] 0000:02:00.0: ttyS6 at MMIO 0x90161000 (irq = 188, base_baud = 3125000) is a 16550A
[    3.181108] 0000:02:00.0: ttyS7 at MMIO 0x90171000 (irq = 189, base_baud = 3125000) is a 16550A
[    3.182604] 0000:02:00.0: ttyS0 at MMIO 0x90181000 (irq = 190, base_baud = 3125000) is a 16550A
[    3.184083] 0000:02:00.0: ttyS1 at MMIO 0x90191000 (irq = 195, base_baud = 3125000) is a 16550A
[    3.185766] Parent device name: 0000:02:00.0
[    3.185770] Child device found: 0000:02:00.0:0
[    3.185775] ptp_ocp 0000:02:00.0: Could not find device ttyS6
[    3.185778] Parent device name: 0000:02:00.0
[    3.185782] Child device found: 0000:02:00.0:0
[    3.185786] ptp_ocp 0000:02:00.0: Could not find device ttyS7
[    3.185788] Parent device name: 0000:02:00.0
[    3.185792] Child device found: 0000:02:00.0:0
[    3.185795] ptp_ocp 0000:02:00.0: Could not find device ttyS0
[    3.185797] Parent device name: 0000:02:00.0
[    3.185801] Child device found: 0000:02:00.0:0
[    3.185805] ptp_ocp 0000:02:00.0: Could not find device ttyS1
[    3.185807] Parent device name: 0000:02:00.0
[    3.185811] Child device found: 0000:02:00.0:0
[    3.185857] ptp_ocp 0000:02:00.0: Version 1.2.0, clock PPS, device ptp3
[    3.185869] ptp_ocp 0000:02:00.0: Time: 1718135008.201674902, in-sync
[    3.185875] ptp_ocp 0000:02:00.0:  GNSS: /dev/ttyS6  @ 115200
[    3.185877] ptp_ocp 0000:02:00.0: GNSS2: /dev/ttyS7  @ 115200
[    3.185877] ptp_ocp 0000:02:00.0:   MAC: /dev/ttyS0  @  57600
[    3.185881] ptp_ocp 0000:02:00.0:  NMEA: /dev/ttyS1  @   9600
```
```
static int
print_child_device(struct device *dev, void *data)
{
    pr_info("Child device found: %s\n", dev_name(dev));
    return 0;
}

void ptp_ocp_link_child(struct ptp_ocp *bp, const char *name, const char *link)
{
    struct device *dev, *child;

    dev = &bp->pdev->dev;
//-------------------------------------
    if (!dev) {
        pr_err("Parent device is NULL\n");
        return;
    }
    pr_info("Parent device name: %s\n", dev_name(dev));

    device_for_each_child(dev, NULL, print_child_device);
//---------------------------------------

    child = device_find_child_by_name(dev, name);
    if (!child) {
        dev_err(dev, "Could not find device %s\n", name);
        return;
    }
```

vvfedorenko, 2024-08-05:
The changes introduced in kernel 6.8 made it impossible to find child tty device as we did before. The changes to the upstream driver are currently under review. In meanwhile you can use Out-Of-Tree driver.

SfCal, 2025-01-07:
Thank you for the update. This worked as expected.

closed 2025-01-07 by SfCal