Time-Appliances-Project/Time-Card #114: Time Stick driver crash while retrieving timestamp

rnijveld, 2024-10-09:
I got a driver crash while working with the time stick given to me during ISPCS2024 on my up-to-date Debian Bookworm machine. It uses kernel 6.1.0-26-amd64. It looks like one of the kernel protection mechanisms is triggered because of a memcpy in `ax_ptp.c:506` (see line 60 in the trace attached below). It appears the kernel isn't happy about the memcpy across multiple fields. Maybe the memcpy call can be replaced by separate calls for each field that needs to be copied from the one struct to the other. Alternatively we memcpy the entire struct and reset the fields that didn't need to be copied?

[time-stick-trace.txt](https://github.com/user-attachments/files/17300287/time-stick-trace.txt)

vvfedorenko, 2024-10-09:
The reason for this error is using CONFIG_FORTIFY_SOURCE=y kernel config option which prevents writing over several fields in structures. Even though the code doesn't corrupt memory, the kernel is pretty strict on the size of data type provided to mem* and str* functions. The easy fix would be to try to explicitly case pointer to `void *`:
`
@@ -503,7 +503,7 @@ static struct _ax_ptp_info *ax_ptp_info_transform(struct ax_device *axdev,
                for (i = 0; i < AX_PTP_HW_QUEUE_SIZE; i++) {
                        memcpy(&temp[i], &_179a_ptp[i], 2);
                        temp[i].sequence_id &= 0xFF;
-                       memcpy(&temp[i].nsec, &_179a_ptp[i].nsec, 10);
+                       memcpy((void *)&temp[i].nsec, (void *)&_179a_ptp[i].nsec, 10);
                }
                memcpy(data, temp, AX_PTP_INFO_SIZE);
                break;
`

I'll push a better solution to the repo to avoid such breaks

vvfedorenko, 2024-10-09:
The fix is pushed to the repo, could you please test it?

closed 2025-01-08 by vvfedorenko