Time-Appliances-Project/Time-Card #115: Inqueries about the Time Card

regymm, 2024-12-08:
Hi, I don't know if this is the right place to ask questions about the Time Card (the mailing list discussion seems quite professional), but I'm working on a project of porting the Time Card FPGA design to [OpenXC7, the open-source FPGA toolchain](https://github.com/openXC7). 

I have some questions about the FPGA design: 

- I saw the clocks for AXI peripherals are two-fold: PTP-related parts use one clock, while system peripherals (UART, etc) use another. Is this only for easier management/coding, or has a performance impact? 
- The DCXO also goes through clocking wizards and BufgMuxs until reaching the PTP clock domain. Will these impact performance? If not, what will be the major time-critical path in the design? 
- Is the I2C bus used on the Time Card on sale? From the schematics, I saw the reference clock crystal oscillator doesn't have any I2C controls. 

Thanks for your attention!

thschaub, 2024-12-09:
Hi 

Your questions are not really issues. Howerer, I can try to answer your questions but I will not teaching you in FPGA design practices.

Regarding your first two questions, they can be addressed by understanding the use of different clock sources for the synchronization part. Certain components are designed to avoid losing lock and therefore use a separate, always-available clock source. Clock Domain Crossing (CDC) is managed through the AXI Interconnects.

In general, most of the design operates at 50MHz, which is not particularly critical. I recommend running the design and analyzing the timing paths to identify any potential critical issues.

As for the I2C usage, this can be clarified by referring to the documentation or building the design itself. You can find the relevant information in the manuals or at the following link:
https://github.com/opencomputeproject/Time-Appliance-Project/tree/master/Time-Card/FPGA/Open-Source/Implementation/Xilinx/TimeCard

Regards

closed 2024-12-09 by thschaub

regymm, 2024-12-10:
Thank you very much for the information!