Time-Appliances-Project/Time-Card #118: Multiple TimeCards synchronization via 1PPS or 10MHz connection

Luka-qw, 2025-04-02:
There are 3 TimeCards. I want to make a stand of 3 TimeCards with a serial connection via 10 MHz SMA and one GPS source. I tried connecting the cards via 10 MHz output of one to the input of the other, but they couldn't synchronize.

Is there any way to synchronize between multiple TimeCards via 1PPS or 10MHz connection?
Exceptionally question is it possible to synchronize without using a gps input?

JulianStj, 2025-05-23:
Thanks for the question. What do you mean by serial connection?

So you want one time card 10MHz output to feed another time card 10MHz input?

Luka-qw, 2025-05-23:
This is exactly what I want. Is it possible?

JulianStj, 2025-05-24:
I think it should be possible, @thschaub  can probably answer better than me. Can you please share what version of Time Card you're using? A picture of the setup would be great too. Thanks!

thschaub, 2025-05-26:
Hi @Luka-qw 

First, some basics: With only the 10MHz signal, you lack phase information because there is no embedded PPS or similar synchronization signal. While using the 10MHz signal may achieve frequency synchronization, it will not ensure phase alignment within the second.

The 10MHz signal via SMA is primarily intended for scenarios where one Time Card has a MAC, and the others either have inexpensive or no oscillators, allowing all cards to use the same stable 10MHz signal. The switch to the 10MHz signal via SMA should happen automatically, but you can verify the selected source using AXI GPIO SEL (GPIO).

To synchronize Time Cards, you always require PPS. This selection should also occur automatically. The presence of PPS can be verified via AXI GPIO SEL (GPIO), and an override of the selection is possible via AXI GPIO SEL (GPIO2).

However, you may still face an issue where your absolute time is not within one second, as only one Time Card seems to have a GNSS receiver providing Time of Day information.

I hope with information you are able to get the cards synchronized.